openrr/urdf-viz #5: Build failed

airfield20, 2018-03-22:
When following your instructions to install I got this output. 
```
Compiling nalgebra v0.13.1
error: failed to run custom build command for `glfw-sys v3.2.2`
process didn't exit successfully: `/tmp/cargo-install.ij9EEj2EbdnX/release/build/glfw-sys-7281d2a56283e13b/build-script-build` (exit code: 101)
--- stdout
running: "cmake" "/home/aaron/.cargo/registry/src/github.com-1ecc6299db9ec823/glfw-sys-3.2.2/." "-DGLFW_BUILD_EXAMPLES=OFF" "-DGLFW_BUILD_TESTS=OFF" "-DGLFW_BUILD_DOCS=OFF" "-DCMAKE_INSTALL_PREFIX=/tmp/cargo-install.ij9EEj2EbdnX/release/build/glfw-sys-93bf9f8db48e2b8c/out" "-DCMAKE_C_FLAGS= -ffunction-sections -fdata-sections -fPIC -m64" "-DCMAKE_C_COMPILER=/usr/bin/cc" "-DCMAKE_CXX_FLAGS= -ffunction-sections -fdata-sections -fPIC -m64" "-DCMAKE_CXX_COMPILER=/usr/bin/c++" "-DCMAKE_BUILD_TYPE=Release"
-- The C compiler identification is Clang 4.0.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Could NOT find Vulkan (missing: VULKAN_LIBRARY VULKAN_INCLUDE_DIR) 
-- Using X11 for window creation
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/x86_64-linux-gnu/libX11.so
-- Configuring incomplete, errors occurred!
See also "/tmp/cargo-install.ij9EEj2EbdnX/release/build/glfw-sys-93bf9f8db48e2b8c/out/build/CMakeFiles/CMakeOutput.log".
See also "/tmp/cargo-install.ij9EEj2EbdnX/release/build/glfw-sys-93bf9f8db48e2b8c/out/build/CMakeFiles/CMakeError.log".

--- stderr
CMake Error at CMakeLists.txt:222 (message):
  The RandR library and headers were not found

thread 'main' panicked at '
command did not execute successfully, got: exit code: 1

build script failed, must exit now', /home/aaron/.cargo/registry/src/github.com-1ecc6299db9ec823/cmake-0.1.29/src/lib.rs:632:5
note: Run with `RUST_BACKTRACE=1` for a backtrace.

warning: build failed, waiting for other jobs to finish...
error: failed to compile `urdf-viz v0.9.0`, intermediate artifacts can be found at `/tmp/cargo-install.ij9EEj2EbdnX`

Caused by:
  build failed

```

OTL, 2018-03-22:
this is repeat of readme, but I want to confirm

Really ran below command?

```
sudo apt-get install cmake xorg-dev libglu1-mesa-dev
```

and tell me your OS.

airfield20, 2018-03-23:
I was missing xorg-dev

closed 2018-03-23 by airfield20

OTL, 2018-03-28:
OK! Good luck!