openrr/urdf-viz #7: index out of bounds

keithmgould, 2018-05-29:
Thank you for making this :) Will post issues as I find them.

I'm using the Mac binary. High Sierra 10.13.4
It worked right away. :)

Issue: when I type '[' the app crashes. Here is the crash with backtrace:

```
thread 'main' panicked at 'index out of bounds: the len is 0 but the index is 0', /Users/travis/build/rust-lang/rust/src/libcore/slice/mod.rs:865:10
stack backtrace:
   0: std::sys::unix::backtrace::tracing::imp::unwind_backtrace
   1: std::sys_common::backtrace::print
   2: std::panicking::default_hook::{{closure}}
   3: std::panicking::default_hook
   4: std::panicking::rust_panic_with_hook
   5: std::panicking::begin_panic_fmt
   6: rust_begin_unwind
   7: core::panicking::panic_fmt
   8: core::panicking::panic_bounds_check
   9: urdf_viz::main
  10: std::rt::lang_start::{{closure}}
  11: std::panicking::try::do_call
  12: __rust_maybe_catch_panic
  13: std::rt::lang_start_internal
  14: main
```

keithmgould, 2018-05-29:
PS: here is URDF file, its pretty simple:

```xml
<?xml version="1.0"?>
<robot name="myfirst">
  <material name="blue">
    <color rgba="0 0 0.8 1"/>
  </material>

  <material name="something">
    <color rgba="1 0.5 0.5 1"/>
  </material>
  
  <material name="black">
    <color rgba="0 0 0 0"/>
  </material>

  <!-- Main Body -->
  <link name="base_link">
    <visual>
      <geometry>
        <box size="0.1016 0.0254 0.6604"/>
      </geometry>
      <origin xyz="0 0 0.3302" />
      <material name="something"/>
    </visual>

    <collision>
      <geometry>
        <box size="0.1016 0.0254 0.6604"/>
      </geometry>
      <origin xyz="0 0 0.3302" />
    </collision>

    <inertial>
      <origin rpy="0 0 0" xyz="0 0 0.2159"/>
      <mass value="1.965" />
      <inertia ixx="0.004878" ixy="-6.2341e-07" ixz="-7.4538e-07" iyy="0.00090164" iyz="-0.00014394" izz="0.0042946"/>
    </inertial>
  </link>

  <!-- Left Wheel Object -->
  <link name="left_wheel">
    <visual>
      <geometry>
        <cylinder length="0.0254" radius=".042"/>
      </geometry>
      <origin rpy="0 1.57 0"/>
      <material name="blue"/>
    </visual>

    <collision>
      <geometry>
        <cylinder length="0.0254" radius=".042"/>
      </geometry>
      <origin rpy="0 1.57 0"/>
    </collision>
  </link>

  <!-- Left Wheel Joint To Body -->
  <joint name="base_to_left_wheel" type="fixed">
    <parent link="base_link"/>
    <child link="left_wheel"/>
    <origin xyz="0.09398 0.0 0.01905" />
  </joint>

  <!-- Right Wheel -->
  <link name="right_wheel">
    <visual>
      <geometry>
        <cylinder length="0.0254" radius=".042"/>
      </geometry>
      <origin rpy="0 1.57 0" />
      <material name="blue"/>
    </visual>

    <collision>
      <geometry>
        <cylinder length="0.5" radius="0.1"/>
      </geometry>
      <origin rpy="0 1.57 0"/>
    </collision>
  </link>

  <!-- Right Wheel Joint to Body -->
  <joint name="base_to_right_wheel" type="fixed">
    <parent link="base_link"/>
    <child link="right_wheel"/>
    <origin xyz="-0.09398 0.0 0.01905" />
  </joint>

</robot>
```

keithmgould, 2018-05-29:
once i changed the joint type from fixed to revolute the app was able to work using `[` I think the issue is when there are only fixed joints.

OTL, 2018-05-29:
Thank you, I'll fix it someday.

commit 74d7ff6d (2018-05-29): Fix bug of out of bounds and rotation direction

Fix #7, #8
closed 2018-05-29 by OTL