openrr/urdf-viz #12: Compilation fails with "underscore lifetimes are unstable (see issue #44524)"

ags-dy, 2018-07-04:
When installing from cargo using:
`cargo install urdf-viz`
 there is a compiler errror:
```

error[E0658]: underscore lifetimes are unstable (see issue #44524)
   --> /home/ags/.cargo/registry/src/github.com-1ecc6299db9ec823/k-0.11.0/src/rctree_links.rs:234:52
    |
234 |     pub fn iter(&self) -> impl Iterator<Item = Ref<'_, Link<T>>> {
    |                                                    ^^

error[E0658]: underscore lifetimes are unstable (see issue #44524)
   --> /home/ags/.cargo/registry/src/github.com-1ecc6299db9ec823/k-0.11.0/src/rctree_links.rs:240:59
    |
240 |     pub fn iter_mut(&self) -> impl Iterator<Item = RefMut<'_, Link<T>>> {
    |                                                           ^^

error[E0658]: underscore lifetimes are unstable (see issue #44524)
   --> /home/ags/.cargo/registry/src/github.com-1ecc6299db9ec823/k-0.11.0/src/rctree_links.rs:250:60
    |
250 |     pub fn iter_movable(&self) -> impl Iterator<Item = Ref<'_, Link<T>>> {
    |                                                            ^^

error[E0658]: underscore lifetimes are unstable (see issue #44524)
   --> /home/ags/.cargo/registry/src/github.com-1ecc6299db9ec823/k-0.11.0/src/rctree_links.rs:256:67
    |
256 |     pub fn iter_movable_mut(&self) -> impl Iterator<Item = RefMut<'_, Link<T>>> {
    |                                                                   ^^

error: aborting due to 4 previous errors

```

The same error can also be reproduced by trying to compile the example from the OTL/gear repository.
`cargo run --release --example reach`

OTL, 2018-07-04:
Please try

```
rustup update
```

if you can.

ags-dy, 2018-07-04:
Ah sorry, I'm still new to rust and that didn't occur to me.
That fixes the problem, thanks.

closed 2018-07-04 by ags-dy

OTL, 2018-07-04:
thank you for trying my code!