openrr/urdf-viz #25: Get method works but set method does not seem to work

prajval10, 2019-07-06:
Probably its a fault on my end but I got the /get_joint_positions method to work and I can visualise it in the browser : 
[image]

However, the set method gives the following error:
```
~$ curl -H "Accept: application/json" -H "Content-type: application/json" -X POST -d '{"names": ["Joint_1", "Joint_2", "Joint_3", "Joint_4", "Joint_5", "Joint_6", "Joint_7"], "positions": [ 0.0, −0.5236, 0.0, −1.0472, 0.0, −1.0472, 0.0]}'  http://127.0.0.1:7777/set_joint_positions | jq
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   258  100   101  100   157    98k   153k --:--:-- --:--:-- --:--:--  251k
{
  "description": "error while parsing the JSON body",
  "cause": {
    "description": "JSON error",
    "cause": null
  }
}
~$ {
>   "is_ok": true,
>   "reason": ""
> 
> 
```

The urdf file is running in another terminal using `urdf-viz <path-to-file>.urdf` command and it works fine.

OTL, 2019-07-08:
I copied your code, and sent it to jq, and it failed. It seems that your json is broken.

```
'{"names": ["Joint_1", "Joint_2", "Joint_3", "Joint_4", "Joint_5", "Joint_6", "Joint_7"], "positions": [ 0.0, −0.5236, 0.0, −1.0472, 0.0, −1.0472, 0.0]}'
```

I replaced your `−` with `-`, it worked.
I don't know about your keyboard, but it is similar to minus, but is not. 

https://www.fileformat.info/info/unicode/char/2212/index.htm

Please use ascii code instead of unicode(−).

prajval10, 2019-07-08:
Ah, classical copy-paste error. Thanks, it works like a charm! 
Closing the issue!

closed 2019-07-08 by prajval10