openrr/urdf-viz #27: thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: File

yapbenzet, 2019-10-04:
I use Ubuntu 16 but i cannot use urdf-viz.. 
I get the following errors when i try to run urdf-viz.

yapbenzet@yapbenzet:~/catkin_ws/src/urdf_tutorial/urdf$ urdf-viz 01-myfirst.urdf 
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: File(Error { repr: Os { code: 13, message: "Permission denied" } })', /checkout/src/libcore/result.rs:860
note: Run with `RUST_BACKTRACE=1` for a backtrace.
yapbenzet@yapbenzet:~/catkin_ws/src/urdf_tutorial/urdf$ export RUST_BACKTRACE=1
yapbenzet@yapbenzet:~/catkin_ws/src/urdf_tutorial/urdf$ urdf-viz 01-myfirst.urdf 
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: File(Error { repr: Os { code: 13, message: "Permission denied" } })', /checkout/src/libcore/result.rs:860
stack backtrace:
   0: std::sys::imp::backtrace::tracing::imp::unwind_backtrace
             at /checkout/src/libstd/sys/unix/backtrace/tracing/gcc_s.rs:49
   1: std::sys_common::backtrace::_print
             at /checkout/src/libstd/sys_common/backtrace.rs:71
   2: std::panicking::default_hook::{{closure}}
             at /checkout/src/libstd/sys_common/backtrace.rs:60
             at /checkout/src/libstd/panicking.rs:355
   3: std::panicking::default_hook
             at /checkout/src/libstd/panicking.rs:371
   4: std::panicking::rust_panic_with_hook
             at /checkout/src/libstd/panicking.rs:549
   5: std::panicking::begin_panic
             at /checkout/src/libstd/panicking.rs:511
   6: std::panicking::begin_panic_fmt
             at /checkout/src/libstd/panicking.rs:495
   7: rust_begin_unwind
             at /checkout/src/libstd/panicking.rs:471
   8: core::panicking::panic_fmt
             at /checkout/src/libcore/panicking.rs:69
   9: core::result::unwrap_failed
  10: urdf_viz::main
  11: __rust_maybe_catch_panic
             at /checkout/src/libpanic_unwind/lib.rs:98
  12: std::rt::lang_start
             at /checkout/src/libstd/panicking.rs:433
             at /checkout/src/libstd/panic.rs:361
             at /checkout/src/libstd/rt.rs:59
  13: __libc_start_main
  14: _start

blooop, 2020-12-14:
I am on ubuntu 18.04 installed via snap and I get this error as well.

OTL, 2020-12-15:
Sorry, snap is not supported.

Could you try other installations?

blooop, 2020-12-15:
If I install via cargo I get a different error.

```
urdf-viz template.urdf 
root [⚓] => /base_link/                                                                                                                                                             
    base_joint_x [↕+X] => /base_linky/                                                                                                                                              
        base_joint_y [↕+Y] => /base_link_rot/                                                                                                                                       
            base_joint_rot [⚙+Z] => /chassis/                                                                                                                                       
                wheelj1 [⚓] => /wheel1/                                                                                                                                             
                wheelj2 [⚓] => /wheel2/                                                                                                                                             
                wheelj3 [⚓] => /wheel3/                                                                                                                                             
                chassis2towerAngle [⚙+Z] => /towerAngleDummy/                                                                                                                       
                    chassis2tower [⚓] => /tower/                                                                                                                                    
                        tower2slider [↕+Z] => /slider/                                                                                                                              
                            slider2rollDummy [⚓] => /shoulder_roll_dummy/                                                                                                           
                                slider2upper_arm [⚙+Y] => /upper_arm/                                                                                                               
                                    elbow_diff [⚙+Z] => /elbow_diff_dummy/                                                                                                          
                                        upper_arm2lower_arm [⚙+Y] => /lower_arm/                                                                                                    
                                            lower_arm2wrist_pitch [⚙+Y] => /wrist_pitch/                                                                                            
                                                wrist_roll2wrist_pitch [⚙+Z] => /wrist3/                                                                                            
                                                    ee_joint [⚓] => /ee/                                                                                                            
                                                                                                                                                                                    
thread 'main' panicked at 'attempted to leave type `platform::platform::x11::util::input::PointerState` uninitialized, which is invalid', /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/core/src/mem/mod.rs:658:9                                                                                                                                             
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace 
```

OTL, 2020-12-15:
I saw that error fist time.

I googled and found a similar issue.

https://github.com/amethyst/amethyst/issues/2524

If you just want to use `urdf-viz`, you can download the binary from [here](https://github.com/OpenRR/urdf-viz/releases)

The file name you have to download is `urdf-viz-v0.19.1-x86_64-unknown-linux-gnu.tar.gz`.

Thank you for trying and be patient!!

blooop, 2020-12-15:
Thanks, the download works. 

Interestingly, if I install the closest version available on cargo:
```cargo install urdf-viz --version 0.19.0```

it comes up with the same pointer error.

both the cargo version and the downloaded version report 0.19.0 if I check the version of the executable (not 0.19.1 as the downloaded version should be).

blooop, 2020-12-15:
I did a binary search on all the available versions available on cargo.

v 0.15.0 fails to compile for other reason, but the oldest compilable version 0.16.0 fails with the pointer error.  If I download 0.16.0 from the github releases page it works fine. 

It looks like it might be an x11 bug.

closed 2021-06-18 by OTL