openrr/urdf-viz #30: thread 'main' panicked at 'assertion failed: `(left == right)`

blooop, 2020-03-31:
This has been happening for a while now on my system.  I forced a reinstall but the problem remains.

This example output of the error was produced on the default urdf from:
https://mymodelrobot.appspot.com/5629499534213120

but the error seems to be independent of the urdf input.

it looks like something to due with rendering.  I am on xubuntu 18.04.   I tried setting RUST_BACKTRACE=1, but that did not seem return the backtrace. 

```
$echo $RUST_BACKTRACE
1

$urdf-viz robot.urdf 
root [⚓] => /base_link/
    cpu_joint [⚓] => /cpu_link/
        upper_base_joint [⚓] => /upper_base_link/
            torso_joint [⚙+Z] => /torso_link/
                head_pan_servo [⚓] => /head_pan_link/
                    head_pan_joint [⚙+Z] => /head_tilt_link/
                        head_tilt_joint [⚙+Y] => /neck_link/
                            neck_joint [⚓] => /head_link/
                                antenna_joint [⚓] => /antenna_link/
                left_shoulder_joint [⚓] => /left_shoulder_link/
                    left_shoulder_forward_joint [⚙+Z] => /left_shoulder_forward_link/
                        left_shoulder_up_joint [⚙+Y] => /left_shoulder_up_link/
                            left_upper_arm_joint [⚓] => /left_upper_arm_link/
                                left_elbow_joint [⚙+Z] => /left_elbow_link/
                                    left_lower_arm_joint [⚓] => /left_lower_arm_link/
                                        left_wrist_joint [⚙+X] => /left_wrist_link/
                                            left_hand_joint [⚓] => /left_hand_link/
                right_shoulder_joint [⚓] => /right_shoulder_link/
                    right_shoulder_forward_joint [⚙+Z] => /right_shoulder_forward_link/
                        right_shoulder_up_joint [⚙+Y] => /right_shoulder_up_link/
                            right_upper_arm_joint [⚓] => /right_upper_arm_link/
                                right_elbow_joint [⚙+Z] => /right_elbow_link/
                                    right_lower_arm_joint [⚓] => /right_lower_arm_link/
                                        right_wrist_joint [⚙+X] => /right_wrist_link/
                                            right_hand_joint [⚓] => /right_hand_link/
    base_laser_joint [⚓] => /base_laser/

end_link_names = ["base_laser_joint", "right_hand_joint", "left_hand_joint", "antenna_joint"]
DoF=11
joint names=["torso_joint", "right_shoulder_forward_joint", "right_shoulder_up_joint", "right_elbow_joint", "right_wrist_joint", "left_shoulder_forward_joint", "left_shoulder_up_joint", "left_elbow_joint", "left_wrist_joint", "head_pan_joint", "head_tilt_joint"]
thread 'main' panicked at 'assertion failed: `(left == right)`
  left: `1281`,
 right: `0`', /home/blooop/.cargo/registry/src/github.com-1ecc6299db9ec823/kiss3d-0.20.1/src/resource/framebuffer_manager.rs:55:17
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace.
```
thanks

OTL, 2020-04-01:
I can visualize the model on Mac and Ubuntu 18.04.
I seems that it is an error of OpenGL (Error code is 1281=0x501)

[image]

It is difficult to debug it for me, and i have no experience of xubuntu.
Sorry for i can't support.

blooop, 2020-04-01:
Ok, thanks.  

I will let you know if I find a solution.

blooop, 2020-05-07:
I still have this issue.  I noticed that kiss3d has a new release now and tried to update urdf-viz to use kiss3d 0.24.1.  I tried to fix the resulting errors, but I don't know how to write rust so I failed.   Maybe updating to the new kiss3d will fix this error?

OTL, 2020-05-10:
I updated kiss3d to latest version in v0.17.0.
(I don't think it will solve the problem)

blooop, 2020-05-11:
You are correct that it didn't fix the problem. Thanks for trying!

closed 2022-07-06 by taiki-e