openrr/urdf-viz #34: Feature: Reload on Button / URDF File Change

blubbi321, 2020-12-21:
Hey there,

thank you for this great project, it is very helpful to me! :)

I have a very small enhancement that would speed up my workflow a lot (and I would add it myself if I knew any Rust).

It would be great if there was a button/keybind to reload the URDF file from disk, so that I have a quick way to see whether my manual tinkering in the file produced the correct results in 3D.

Even better would be a file watch to automatically update the visualization, but I guess that is way more effort ;)

Thanks a lot for considering it!

blooop, 2020-12-21:
As a workaround I find the opening urdf-viz from the console and pressing ctrl-c, up arrow and enter to reload urdf-viz with the same command is pretty quick when I’m making lots of changes to a Urdf.

OTL, 2020-12-22:
'L' key is to reload the urdf from file. (See the README.md carefully)

(I'm sorry if i missed your point.)

It's not easy to add buttons.
It might be possible to automatically reload the file, and it seems to be easy.
Thank you for nice suggestions!

closed 2021-02-02 by OTL