openrr/urdf-viz #56: Joint Id +- not working with []

hunter-digi-ace, 2021-04-12:
Hello, 
Thanks for your work.

I can use all the functionalities showed on the readme and on screen instructions but the one tho change the joint id.
the buttons [] are perfectly working on my keyboard, but not on the urdf-viz.
That is caused by the fact that I have an ISO keyboard that don't have the keys [] but the (alt-gr + `) and (alt-gr + +) combinations. (I have already tested it with my keyboard in US english layout)

Like this one:  [image](https://ae01.alicdn.com/kf/H32ec617844884fcfbf5755d0ba51cf15J/Teclas-ANSI-ISO-ES-europeo-OEM-PBT-para-interruptores-Cherry-MX-de-teclado-mec-nico-124.jpg)

Can add more key options to change the joint id?

Thanks in advance!

OTL, 2021-04-13:
Thank you for the nice suggestion!
Do you have an idea to solve that issue? (Which key do you want to use (will be added), or make it configurable?)

hunter-digi-ace, 2021-04-13:
Hello again, if it could be configured with a small .txt that was loaded when executing the program it would be a good solution, in case there are more problems with other keyboard layouts, but a perhaps faster solution would be to put generic keyboard keys such as numbers or more usual letters.

If you want to continue maintaining the same configuration as until now, I would put a couple more lines in urdf-viz.rs after line 298, giving the same use to two other keys, for example the numbers _1 and _2 (from the top row keys) so all keyboards could use it, I think.

PR #63 (2021-04-16): Make `o` and `p` alias for `[` and `]`
On the US keyboard layout, op is just to the left of [], so hopefully, this does not worse the usability much.

Closes #56
cc @hunter-digi-ace
closed 2021-04-16 by taiki-e