openrr/urdf-viz #143: cargo install error

GuancongLuo, 2022-04-20:
Compiling k v0.28.0
error: there is no argument named `node`
  --> /home/clarence/.cargo/registry/src/github.com-1ecc6299db9ec823/k-0.28.0/src/chain.rs:99:28
   |
99 |             writeln!(f, "{}{node}", "    ".repeat(level))?;
   |                            ^^^^^^

error: there is no argument named `parent_node`
   --> /home/clarence/.cargo/registry/src/github.com-1ecc6299db9ec823/k-0.28.0/src/urdf.rs:290:45
    |
290 |                         info!("set parent = {parent_node}, child = {child_node}");
    |                                             ^^^^^^^^^^^^^

error: there is no argument named `child_node`
   --> /home/clarence/.cargo/registry/src/github.com-1ecc6299db9ec823/k-0.28.0/src/urdf.rs:290:68
    |
290 |                         info!("set parent = {parent_node}, child = {child_node}");
    |                                                                    ^^^^^^^^^^^^

error: there is no argument named `root_node`
   --> /home/clarence/.cargo/registry/src/github.com-1ecc6299db9ec823/k-0.28.0/src/urdf.rs:316:33
    |
316 |             info!("set parent = {root_node}, child = {rjn}");
    |                                 ^^^^^^^^^^^

error: there is no argument named `rjn`
   --> /home/clarence/.cargo/registry/src/github.com-1ecc6299db9ec823/k-0.28.0/src/urdf.rs:316:54
    |
316 |             info!("set parent = {root_node}, child = {rjn}");
    |                                                      ^^^^^

error: could not compile `k` due to 5 previous errors
warning: build failed, waiting for other jobs to finish...
error: failed to compile `urdf-viz v0.36.0`, intermediate artifacts can be found at `/tmp/cargo-installL8OaLi`

Caused by:
  build failed

[image]

OTL, 2022-04-20:
It seems that you need to update your rust version to the latest.

taiki-e, 2022-05-09:
The current urdf-viz requires Rust 1.58+.

closed 2022-05-09 by taiki-e