openrr/urdf-viz #169: libassimp.so.5 not found error

nicholscrawford, 2022-08-29:
Attempting to install, not a previous rust user. Installed rust, and then tried:
`cargo install urdf-viz`
and when trying to run 
`urdf-viz`
I get
`urdf-viz: error while loading shared libraries: libassimp.so.5: cannot open shared object file: No such file or directory`
though I have ros installed, I tried installing the pre-req packages and recompiling, and no change. Tried to use the more bare-bones install, and couldn't get that to work just by following readme.
```
$ cargo build --no-default-features
error: could not find `Cargo.toml` in `/home/nichols` or any parent directory
```

OTL, 2022-08-30:
> urdf-viz: error while loading shared libraries: libassimp.so.5: cannot open shared object file: No such file or directory

Could you tell us your OS and its version? It will help us a lot.

> $ cargo build --no-default-features
> error: could not find `Cargo.toml` in `/home/nichols` or any parent directory

It seems that you didn't run the command in urdf-viz/ directory.
How about trying below?

```bash
cd urdf-viz/
cargo install --no-default-features --path .
```

taiki-e, 2022-10-19:
> $ cargo build --no-default-features

This is a mistake in the current readme (#174). Could you try `cargo install urdf-viz --no-default-features`?

nicholscrawford, 2022-10-21:
That works, thank you!

closed 2022-10-21 by nicholscrawford