openrr/urdf-viz #180: HTTP keep-alive

emileb, 2022-11-08:
Does the HTTP interface use 'keep-alive' (Sorry I tried looking in to the code but I do not know Rust)? I am trying to make the set_joint_positions HTTP post as performant as possible.
Thanks!

OTL, 2022-11-09:
Hi, thank you for a nice issue!
@taiki-e Do you know?

taiki-e, 2022-11-09:
If you are using HTTP/1.1, keep-alive should be enabled by default; for HTTP/1.0, `Connection: Keep-Alive` header is required.

---

HTTP/1.1:

```console
$ curl http://127.0.0.1:7777/get_joint_positions http://127.0.0.1:7777/get_joint_positions -v
*   Trying 127.0.0.1:7777...
* Connected to 127.0.0.1 (127.0.0.1) port 7777 (#0)
> GET /get_joint_positions HTTP/1.1
> Host: 127.0.0.1:7777
> User-Agent: curl/7.86.0
> Accept: */*
> 
* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< content-type: application/json
< content-length: 273
< date: Wed, 09 Nov 2022 03:26:32 GMT
< 
* Connection #0 to host 127.0.0.1 left intact
{"names":["r_shoulder_yaw","r_shoulder_pitch","r_shoulder_roll","r_elbow_pitch","r_wrist_yaw","r_wrist_pitch","l_shoulder_yaw","l_shoulder_pitch","l_shoulder_roll","l_elbow_pitch","l_wrist_yaw","l_wrist_pitch"],"positions":[0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0]}* Found bundle for host: 0x6000010c80f0 [serially]
* Can not multiplex, even if we wanted to
* Re-using existing connection #0 with host 127.0.0.1
* Connected to 127.0.0.1 (127.0.0.1) port 7777 (#0)
> GET /get_joint_positions HTTP/1.1
> Host: 127.0.0.1:7777
> User-Agent: curl/7.86.0
> Accept: */*
> 
* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< content-type: application/json
< content-length: 273
< date: Wed, 09 Nov 2022 03:26:32 GMT
< 
* Connection #0 to host 127.0.0.1 left intact
{"names":["r_shoulder_yaw","r_shoulder_pitch","r_shoulder_roll","r_elbow_pitch","r_wrist_yaw","r_wrist_pitch","l_shoulder_yaw","l_shoulder_pitch","l_shoulder_roll","l_elbow_pitch","l_wrist_yaw","l_wrist_pitch"],"positions":[0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0]}
```

See "Connection #0 to host 127.0.0.1 left intact" and "Re-using existing connection #0 with host 127.0.0.1" -- keep-alive is enabled.

---

HTTP/1.0:

```console
$ curl http://127.0.0.1:7777/get_joint_positions http://127.0.0.1:7777/get_joint_positions -v --http1.0
*   Trying 127.0.0.1:7777...
* Connected to 127.0.0.1 (127.0.0.1) port 7777 (#0)
> GET /get_joint_positions HTTP/1.0
> Host: 127.0.0.1:7777
> User-Agent: curl/7.86.0
> Accept: */*
> 
* Mark bundle as not supporting multiuse
* HTTP 1.0, assume close after body
< HTTP/1.0 200 OK
< content-type: application/json
< content-length: 273
< date: Wed, 09 Nov 2022 03:31:31 GMT
< 
* Closing connection 0
{"names":["r_shoulder_yaw","r_shoulder_pitch","r_shoulder_roll","r_elbow_pitch","r_wrist_yaw","r_wrist_pitch","l_shoulder_yaw","l_shoulder_pitch","l_shoulder_roll","l_elbow_pitch","l_wrist_yaw","l_wrist_pitch"],"positions":[0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0]}* Hostname 127.0.0.1 was found in DNS cache
*   Trying 127.0.0.1:7777...
* Connected to 127.0.0.1 (127.0.0.1) port 7777 (#1)
> GET /get_joint_positions HTTP/1.0
> Host: 127.0.0.1:7777
> User-Agent: curl/7.86.0
> Accept: */*
> 
* Mark bundle as not supporting multiuse
* HTTP 1.0, assume close after body
< HTTP/1.0 200 OK
< content-type: application/json
< content-length: 273
< date: Wed, 09 Nov 2022 03:31:31 GMT
< 
* Closing connection 1
{"names":["r_shoulder_yaw","r_shoulder_pitch","r_shoulder_roll","r_elbow_pitch","r_wrist_yaw","r_wrist_pitch","l_shoulder_yaw","l_shoulder_pitch","l_shoulder_roll","l_elbow_pitch","l_wrist_yaw","l_wrist_pitch"],"positions":[0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0]}
```

See "Closing connection 0" and "Closing connection 1" -- keep-alive is not enabled.

closed 2022-11-09 by taiki-e

emileb, 2022-11-09:
Ah awesome, thanks for the info and great program 👍