openrr/urdf-viz #192: feature request: change size of link frame and toggle short cuts menu

aheadofkym, 2023-01-23:
Hello,

This is a great tool, thank you for making it.

2 feature requests:
Could i request a way to change the size/scale of teh frame axis when on? 
For small parts like single arms, having the link frames on to double check where the joints are, sometimes the link frames are larger then all the parts.

[image]

Could there be a way to toggle off the shortcuts menu?

Thanks
Kym

OTL, 2023-01-24:
Hi, nice suggestion!
I completely understand the first request. I'll do it soon.

> Could there be a way to toggle off the shortcuts menu?

Do you mean that you want to remove the text on the screen?

kaaatsu32329, 2023-01-25:
@aheadofkym
Thank you for nice suggestions! I updated `urdf-viz` to implement requiested features. How do you like the added features in this PR (#194)?

aheadofkym, 2023-01-26:
@kaaatsu32329 , thats fantastic, thankyou.

closed 2023-01-27 by kaaatsu32329