openrr/urdf-viz #193: feature request: option to set the joint angle / movement value

aheadofkym, 2023-01-24:
Hello,

I was wondering if it's possible to get an option on the command line or something that allows us the configure the joint angle.
Right now it's set to  +/-0.1.
When using a prismatic joint, I suspect this becomes +/-100mm of movement.  

I'm working with a tiny arm, and each side of the gripper only moves 54mm.

Example joint : 
`<joint name="link8-link9" type="prismatic">
  <origin xyz="0.0313 -0.0325 0.0071" rpy="0 0 0"/>
  <parent link="link8"/>
  <child link="link9"/>
  <axis xyz="0.0 1.0 0.0"/>
  <limit upper="0.054" lower="0.054" effort="100" velocity="100"/>
</joint>`

It would be nice to be able to use the tool to test small robots too.

Cheers
Kym

OTL, 2023-01-25:
Do you have any idea to set the value?
How about `(max - min) / N` ?

aheadofkym, 2023-01-26:
Thanks for the reply,

Adding it to the CLI is fine, `-j / --joint-angle 0.1` . I'm not sure how useful it would be to set the min/max independently, so one value is enough.

Cheers
Kym

abrzozowski, 2023-04-12:
Hi guys. Is this task still pending or can we close it because it was completed in https://github.com/openrr/urdf-viz/pull/200?

OTL, 2023-04-13:
I think we can close this, but we need a release.

OTL, 2023-04-13:
@taiki-e Can you release this? v0.42.0?

OTL, 2023-04-19:
released v0.42.0

closed 2023-04-19 by OTL