openrr/urdf-viz #204: Error when running pre-built Windows executable

rserban, 2023-04-11:
I downloaded the pre-built **Windows** binaries for v0.41.0.  `urdef-viz.exe` works fine with the `sample.urdf` provided. However, running it with any URDF input file that uses meshes (STL), I get the following error:

```
thread 'main' panicked at 'rospack find failed: Error { kind: NotFound, message: "program not found" }', C:\Users\runneradmin\.cargo\registry\src\github.com-1ecc6299db9ec823\urdf-rs-0.6.7\src\utils.rs:44:10
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

Setting RUST_BACKTRACE to `full` gives the following:

```
thread 'main' panicked at 'rospack find failed: Error { kind: NotFound, message: "program not found" }', C:\Users\runneradmin\.cargo\registry\src\github.com-1ecc6299db9ec823\urdf-rs-0.6.7\src\utils.rs:44:10
stack backtrace:
   0:     0x7ff7e8e12857 - Assimp::Importer::Pimpl
   1:     0x7ff7e8b8e70b - Assimp::BaseProcess::RequireVerboseFormat
   2:     0x7ff7e8df1cb1 - Assimp::Importer::Pimpl
   3:     0x7ff7e8e145d2 - Assimp::Importer::Pimpl
   4:     0x7ff7e8e14f2e - Assimp::Importer::Pimpl
   5:     0x7ff7e8e14ae5 - Assimp::Importer::Pimpl
   6:     0x7ff7e8e14a2f - Assimp::Importer::Pimpl
   7:     0x7ff7e8e14a04 - Assimp::Importer::Pimpl
   8:     0x7ff7e9310025 - ODDLParser::OpenDDLExport::writeValueType
   9:     0x7ff7e93102e3 - ODDLParser::OpenDDLExport::writeValueType
  10:     0x7ff7e8f39c4f - Assimp::ScenePreprocessor::SetScene
  11:     0x7ff7e8eb9381 - Assimp::ScenePreprocessor::SetScene
  12:     0x7ff7e8ec6368 - Assimp::ScenePreprocessor::SetScene
  13:     0x7ff7e8f6d5aa - Assimp::ScenePreprocessor::SetScene
  14:     0x7ff7e8f9da76 - Assimp::ScenePreprocessor::SetScene
  15:     0x7ff7e8f7e046 - Assimp::ScenePreprocessor::SetScene
  16:     0x7ff7e8f9d32a - Assimp::ScenePreprocessor::SetScene
  17:     0x7ff7e930d03c - ODDLParser::OpenDDLExport::writeValueType
  18:     0x7ff8a0dc269d - BaseThreadInitThunk
  19:     0x7ff8a132a9f8 - RtlUserThreadStart
```

What am I missing?  Thanks.

rserban, 2023-04-11:
Works fine if geometry is provided as Wavefront OBJ files

OTL, 2023-04-12:
Could you tell us what is your path of mesh file?
For example, `filename ="./aaa/bbb.stl"` or `filename="package://aaa/bbb.stl"` and so on.

rserban, 2023-04-14:
Sorry for the delay (GitHub email notifications were in the Junk folder).
The URDF file specifies mesh filenames using the 2nd version above (`filename="package://aaa/bbb.stl`).

rserban, 2023-04-15:
I tested a file that uses paths to the mesh files in the 1st form above (`filename ="./aaa/bbb.stl`) and that worked fine.

taiki-e, 2023-10-18:
Closing as a duplicate of #26.

closed 2023-10-18 by taiki-e