openrr/urdf-viz #253: Failed to run on Ubuntu 24.04

OTL, 2024-07-14:
```bash
$ ./target/release/urdf-viz sample.urdf 
2024-07-14T06:15:12.515949Z  INFO k::urdf: link=root
2024-07-14T06:15:12.515973Z  INFO k::urdf: root=root
2024-07-14T06:15:12.515986Z  INFO k::urdf: link=l_shoulder1
2024-07-14T06:15:12.515990Z  INFO k::urdf: set parent = l_shoulder_yaw [⚙+Z], child = l_shoulder_pitch [⚙+Y]
2024-07-14T06:15:12.516000Z  INFO k::urdf: link=l_shoulder2
2024-07-14T06:15:12.516003Z  INFO k::urdf: set parent = l_shoulder_pitch [⚙+Y], child = l_shoulder_roll [⚙+X]
2024-07-14T06:15:12.516007Z  INFO k::urdf: link=l_shoulder3
2024-07-14T06:15:12.516009Z  INFO k::urdf: set parent = l_shoulder_roll [⚙+X], child = l_elbow_pitch [⚙+Y]
2024-07-14T06:15:12.516013Z  INFO k::urdf: link=l_elbow1
2024-07-14T06:15:12.516015Z  INFO k::urdf: set parent = l_elbow_pitch [⚙+Y], child = l_wrist_yaw [⚙+Z]
2024-07-14T06:15:12.516018Z  INFO k::urdf: link=l_wrist1
2024-07-14T06:15:12.516021Z  INFO k::urdf: set parent = l_wrist_yaw [⚙+Z], child = l_wrist_pitch [⚙+Y]
2024-07-14T06:15:12.516024Z  INFO k::urdf: link=l_wrist2
2024-07-14T06:15:12.516027Z  INFO k::urdf: link=r_shoulder1
2024-07-14T06:15:12.516030Z  INFO k::urdf: set parent = r_shoulder_yaw [⚙+Z], child = r_shoulder_pitch [⚙+Y]
2024-07-14T06:15:12.516033Z  INFO k::urdf: link=r_shoulder2
2024-07-14T06:15:12.516036Z  INFO k::urdf: set parent = r_shoulder_pitch [⚙+Y], child = r_shoulder_roll [⚙+X]
2024-07-14T06:15:12.516039Z  INFO k::urdf: link=r_shoulder3
2024-07-14T06:15:12.516041Z  INFO k::urdf: set parent = r_shoulder_roll [⚙+X], child = r_elbow_pitch [⚙+Y]
2024-07-14T06:15:12.516047Z  INFO k::urdf: link=r_elbow1
2024-07-14T06:15:12.516051Z  INFO k::urdf: set parent = r_elbow_pitch [⚙+Y], child = r_wrist_yaw [⚙+Z]
2024-07-14T06:15:12.516055Z  INFO k::urdf: link=r_wrist1
2024-07-14T06:15:12.516057Z  INFO k::urdf: set parent = r_wrist_yaw [⚙+Z], child = r_wrist_pitch [⚙+Y]
2024-07-14T06:15:12.516060Z  INFO k::urdf: link=r_wrist2
2024-07-14T06:15:12.516064Z  INFO k::urdf: set parent = root [⚓] => /root/, child = l_shoulder_yaw [⚙+Z] => /l_shoulder1/
2024-07-14T06:15:12.516075Z  INFO k::urdf: set parent = root [⚓] => /root/, child = r_shoulder_yaw [⚙+Z] => /r_shoulder1/
root [⚓] => /root/
    l_shoulder_yaw [⚙+Z] => /l_shoulder1/
        l_shoulder_pitch [⚙+Y] => /l_shoulder2/
            l_shoulder_roll [⚙+X] => /l_shoulder3/
                l_elbow_pitch [⚙+Y] => /l_elbow1/
                    l_wrist_yaw [⚙+Z] => /l_wrist1/
                        l_wrist_pitch [⚙+Y] => /l_wrist2/
    r_shoulder_yaw [⚙+Z] => /r_shoulder1/
        r_shoulder_pitch [⚙+Y] => /r_shoulder2/
            r_shoulder_roll [⚙+X] => /r_shoulder3/
                r_elbow_pitch [⚙+Y] => /r_elbow1/
                    r_wrist_yaw [⚙+Z] => /r_wrist1/
                        r_wrist_pitch [⚙+Y] => /r_wrist2/

interface 'wl_surface' has no event 2
2024-07-14T06:15:12.543016Z ERROR smithay_client_toolkit::window::concept_frame: No font could be found    
thread 'main' panicked at /home/ogura/.cargo/registry/src/index.crates.io-6f17d22bba15001f/kiss3d-0.35.0/src/window/gl_canvas.rs:80:14:
called `Result::unwrap()` on an `Err` value: OsError("eglInitialize failed")
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: queue 0x626676cdeae0 destroyed while proxies still attached:
  wl_buffer@40 still attached
  wl_buffer@39 still attached
  wl_buffer@38 still attached
  wl_buffer@37 still attached
  wl_buffer@36 still attached
  wl_subsurface@30 still attached
  wl_surface@29 still attached
  wl_subsurface@28 still attached
  wl_surface@27 still attached
  wl_subsurface@26 still attached
  wl_surface@25 still attached
  wl_subsurface@24 still attached
  wl_surface@23 still attached
  wl_subsurface@22 still attached
  wl_surface@21 still attached
  xdg_wm_base@18 still attached
  wl_surface@17 still attached
  zwp_text_input_manager_v3@11 still attached
  zwp_pointer_constraints_v1@10 still attached
  zwp_relative_pointer_manager_v1@9 still attached
  wl_seat@8 still attached
  wl_subcompositor@7 still attached
  wl_output@6 still attached
  wl_shm@5 still attached
  wl_compositor@4 still attached
  wl_registry@2 still attached
```

taiki-e, 2024-07-14:
> ```shell
> smithay_client_toolkit::window::concept_frame: No font could be found 
> ```

Either you are not running https://github.com/openrr/urdf-viz#pre-requirements-for-build or that instruction is out of date for the latest ubuntu and needs to be updated?

OTL, 2024-07-14:
I did

`sudo apt-get install cmake xorg-dev libglu1-mesa-dev`

It might be the problem of the new laptop hardware.

rainycocoboo, 2024-09-19:
Excuse me,I have the same problem on Ubuntu 22.04,have you solved this problem?

foxton9, 2024-11-21:
The same problem occurred on an Ubuntu 24.04 virtual machine, and forcing the application to use X11 served as a workaround for me.

```sh
WINIT_UNIX_BACKEND=x11 urdf-viz URDF_FILE.urdf
```

matiasg, 2025-03-06:
> The same problem occurred on an Ubuntu 24.04 virtual machine, and forcing the application to use X11 served as a workaround for me.
> 
> WINIT_UNIX_BACKEND=x11 urdf-viz URDF_FILE.urdf

This worked for me in Arch Linux too. Should it be mentioned in the README?