openrr/urdf-viz #256: run ROS urdf example but different visualization result

Jackiezhou233, 2024-07-29:
I run the following example using roslaunch urdf_tutorial display.launch model:=urdf/04-materials.urdf
https://wiki.ros.org/urdf/Tutorials/Building%20a%20Visual%20Robot%20Model%20with%20URDF%20from%20Scratch#Origins
the visualization is: 
[image]
But when I use the same urdf file through urdf-viz, it turn out to be:
[image]

the urdf file is :
`<?xml version="1.0"?>
<robot name="origins">
  <link name="base_link">
    <visual>
      <geometry>
        <cylinder length="0.6" radius="0.2"/>
      </geometry>
    </visual>
  </link>

  <link name="right_leg">
    <visual>
      <geometry>
        <box size="0.6 0.1 0.2"/>
      </geometry>
      <origin rpy="0 1.57075 0" xyz="0 0 -0.3"/>
    </visual>
  </link>

  <joint name="base_to_right_leg" type="fixed">
    <parent link="base_link"/>
    <child link="right_leg"/>
    <origin xyz="0 -0.22 0.25"/>
  </joint>

</robot>`

Why?? the tf transform problem?

taiki-e, 2024-09-03:
This is due to one of the regressions introduced in urdf-rs 0.7 (https://github.com/openrr/urdf-rs/pull/64) that the default value was accidentally changed.

I confirmed that this will be fixed by https://github.com/openrr/urdf-rs/pull/107.

[image]

closed 2024-09-04 by taiki-e

taiki-e, 2024-09-12:
Fixed in 0.46.0.