openrr/urdf-viz #262: Incorrect mesh visualization

Divelix, 2024-09-05:
I tried to visualize Zed X camera from [original repo](https://github.com/stereolabs/zed-ros2-interfaces/tree/main/meshes) and there is how `urdf-viz` displays it:
[image]

This problem appears in other meshes too, so I conclude that something is wrong with `urdf-viz` renderer.

Divelix, 2024-09-05:
There is how `zedx.stl` looks like in `meshlab`:
[image]

taiki-e, 2024-09-06:
Thanks for the report! We recently fixed some bugs in urdf parser and mesh parser, so may be fixed in https://github.com/openrr/urdf-viz/pull/263 which update them.

r7vme, 2024-10-31:
I use binary v0.46.0 version and have similar problem

[image]

taiki-e, 2024-11-06:
@r7vme Could you provide the information about how to reproduce the problem?